bdring/Grbl_Pen_Servo #1: Servo rotates randomly

roydekleijn, 2019-03-25:
Hi Bart,

This looks promising for my project! However, I connected the servo to pin 11 (Z-) of the CNC shield V4 and it rotates randomly.

I'm guessing this is not the correct pin ??

Can you please help? I have an Arduino nano, cnc shield V4 and tower pro MG90s servo.

Thanks,
Roy

bdring, 2019-03-25:
I'll look into that as soon as I get to my workbench

bdring, 2019-03-26:
Can you send me a link to the exact shield you are using? I need to check the pinout.

roydekleijn, 2019-03-26:
Hi,

The exact board is
https://www.instructables.com/id/How-to-Use-the-CNC-V4-Board-despite-Its-quirks/

The CoreXY movement is working fine, but the servo doesn't move...

Roy

bdring, 2019-03-26:
You should connect the servo signal pin to the z endstop pin PortB3 or digital 11.

Send a picture of your connection.

roydekleijn, 2019-03-26:
Herewith the picture:
[image]

[image]

And grbl settings:
[image]

Which command do you expect the servo to move?

Do I need to change some value in config.h ?

Many thanks in advance,
Roy

bdring, 2019-03-26:
Move Z to move servo 

G0 Z-5
G0 Z5

roydekleijn, 2019-03-26:
Really strange! it doesn't move :(

Is it needed to update some value in config.h (or any other file) ?

bdring, 2019-03-26:
I just tested it on my controller and it runs fine.

https://www.tindie.com/products/33366583/penlaser-bot-controller/

Your Z rate is rather slow. It would take 1 minute to do the moves I suggested. Try changing it to 500

roydekleijn, 2019-03-26:
[image]

I see D11 is connected to both Z+ and Z- pins. Does it matter to which pin I connect the servo ?

bdring, 2019-03-26:
I does not matter which one, but it must be the inner row where the trace goes to and not the row along the edge of the board.

roydekleijn, 2019-03-26:
Ok, it's connected to Z- pin

if I execute `G0 Z5` I feel some (and hear) some resistance on the steppers... (does that clarify something ?

Again, normal X / Y movement works correct.

bdring, 2019-03-26:
That is normal

You have a step idle delay of 25msec  ($1=25). This turns of the motors after 25msec of no movement. The Z move re-energizes them unit 25msec after the z move is done.

- Are you sure you have the servo hooked up right? 5V, Pulse and Ground
- Are you sure the servo works
- Have you tried the basic servo sweep example sketch with pin 11?

roydekleijn, 2019-03-27:
Thanks, I tried the sweep sketch and it works! so it must be related to GRBL config.h or the GRBL settings... What is the best approach to tackle this issue ?

bdring, 2019-03-27:
I don't know. Have you used other forks of Grbl before? Are you 100% sure you are using this code? Grbl is installed as a library, so it is easy to make a mistake.

Try sending $rst=* to reset all settings.

I did a fresh install on my board and it works fine, Others have used it as well.

bdring, 2019-03-27:
Here is my blog post on the method.

http://www.buildlog.net/blog/2017/08/using-grbls-spindle-pwm-to-control-a-servo/

roydekleijn, 2019-03-27:
Sure, will try this tonight

I will remove it from the library folder and download it again.

Does it have to do with $23 ? which is set to 4 in my case.

bdring, 2019-03-27:
I don't think so. Until you get it working, you might change $22 to 0. That disables homing. Grbl comes up in Idle, rather than Alarm

roydekleijn, 2019-03-27:
Honestly I don't know what went wrong, but it's working now!

- I did a clean clone
- I added the library to Arduino IDE
- I enable `#define COREXY`
- I tried to upload the firmware to my Arduino nano, but didn't work because of the size.
- I disabled the following:
#define RESTORE_OVERRIDES_AFTER_PROGRAM_END // Default enabled. Comment to disable.
#define REPORT_FIELD_WORK_COORD_OFFSET // Default enabled. Comment to disable.
#define REPORT_FIELD_OVERRIDES // Default enabled. Comment to disable.
#define REPORT_FIELD_LINE_NUMBERS // Default enabled. Comment to disable.

- I tried to upload again, this time it was 'small' enough.

bdring, 2019-03-27:
OK, Glad to hear that.

[Read this regarding the program size](http://www.buildlog.net/blog/2017/09/getting-more-program-space-on-an-arduino-nano/).

If you believe my firmware, support or blog posts have helped, please consider a donation via the link at the [bottom of this page](https://github.com/bdring/Grbl_Pen_Servo).

roydekleijn, 2019-03-28:
I noticed the "soft limits" are not working. Does it have to do with the commented line?

roydekleijn, 2019-03-28:
Sure will donate as you were so kind to help me out...

One more thing. I enabled "soft limit" and noticed that the pencil doesn't go up or down. (also, I like to start it in the "up" position)

These are my GRBL settings:
[image]

bdring, 2019-03-28:
I just tested soft limits for you and they are working on my setup. 

Soft limits and Z are tricky because that is a not a homed axis. Grbl typically homes into negative machine space so any positive move in Z will exceed the soft limits. You could set a work offset in G54 so you can move above and below Z0 in that work coordinate system. 

[Here is a video about work offsets I made](https://www.youtube.com/watch?v=fGtbkVJBXyE).

If you understand them and how you are generating the gcode, you might be able to use that to make the soft limits work and get the pen to go up at startup.

roydekleijn, 2019-03-30:
I still try to get the soft limits working:

GRBL settings:

```
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 18    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 2500.000    (Homing locate feed rate, mm/min)
$25 = 5000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 50.000    (X-axis travel resolution, step/mm)
$101 = 50.000    (Y-axis travel resolution, step/mm)
$102 = 10.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 150.000    (X-axis maximum travel, millimeters)
$131 = 150.000    (Y-axis maximum travel, millimeters)
$132 = 10.000    (Z-axis maximum travel, millimeters)
```

Ouput:

```
$H
ok
>>> ?
[verbose]<Idle|WPos:-2.000,-2.000,0.000>
ok
>>> G10 P0 L20 X0 Y0 Z0
ok
>>> ?
[verbose]<Idle|WPos:0.000,0.000,0.000>
ok
>>> G0 Z5
An error was detected while sending 'G0 Z5': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused.
[MSG:Reset to continue]
```

The pen goes up/down when I disable 'soft limit', however I really like to us soft limiting. Do you have an idea?

roydekleijn, 2019-03-30:
By doing:

```
>>> $N
$N0=G54Z5
```

it seems to work, it starts in upward position. But till when will this setting set?

roydekleijn, 2019-03-30:
Apparently not.. Again I get an Alarm state after homing

```
>>> ?
[verbose]<Alarm|WPos:0.000,0.000,-5.000>
ok
>>> $#
[G54:0.000,0.000,5.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
>>> $H
(ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked.(ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked.
[MSG:Reset to continue]

```

bdring, 2019-03-30:
I am on vacation and not able to test a solution for you.  I suggest not using soft limits. 

Your previous $N statement is not right. Typically offsets are set with the G10 command. The offset only needs to be applied once and will be remembered even after the arduino is turned off

Again.....I suggest not using soft limits until after you have experience with the machine. 

By default grbl creates negative machine space. If your max Z is 30, your valid range is 0 to -29. If you try to move to +1, if is outside of the limits.

The G54 to G59 work coordinate systems allow you to create positive work coordinates inside the negative machine coordinates.

If is very robust, but complicated for new users. 

This is not an issue with my fork or regular Grbl. It is basic CNC and NIST gcode

roydekleijn, 2019-03-30:
Ok thanks for the detailed explanation. I will disable soft limit for now.

Enjoy your vacation!

closed 2019-03-30 by roydekleijn

roydekleijn, 2019-04-03:
Maybe one more thing, which might be related to this fork or my GRBL settings 

- I noticed that moving the pencil up/down is a bit slow.  

How can I improve the speed of moving the pen up/down ?

bdring, 2019-04-03:
The servo will rotate as fast as it can as soon as it sees a positive Z
move. The next gcode will not begin until that "virtual" Z move is
complete. That might be the slow down you are having.

The length of the move plus the speed and acceleration of the Z axis
determine how long that virtual move takes. Shorten the move and speed up
those settings (via $$ menu). Be careful that the virtual move does not
complete faster than the servo can move or the X,Y move may start before
the pen has moved completely.

JamesCWhite, 2019-10-13:
For anyone who is curious, you can get the servo Z axis to play nicely with grbl soft limits by using work coordinates as Bart eluded to. I'm using this on a 'stock' midtbot esp32 with a servo Z axis mapped to 0-5mm.

After homing, issue a G0 Z-5
The servo should not move.

Follow it with 
G10 L20 P1 Z0

After this, your gcode can issue a Z5 pen up and still be in (at the edge of) the machine coordinates negative space, allowing the soft limits to work. After that point it's stored in eeprom so you don't need to change it again unless you re-zero the z axis.

mathematicalmichael, 2020-04-18:
> Honestly I don't know what went wrong, but it's working now!
> 
> * I did a clean clone
> * I added the library to Arduino IDE
> * I enable `#define COREXY`  (in config.h)
> * I tried to upload the firmware to my Arduino nano, but didn't work because of the size.
> * I disabled the following (in config.h):
>   `#define RESTORE_OVERRIDES_AFTER_PROGRAM_END // Default enabled. Comment to disable.`
>   `#define REPORT_FIELD_WORK_COORD_OFFSET // Default enabled. Comment to disable.`
>   `#define REPORT_FIELD_OVERRIDES // Default enabled. Comment to disable.`
>   `#define REPORT_FIELD_LINE_NUMBERS // Default enabled. Comment to disable.`
> * I tried to upload again, this time it was 'small' enough.

SIR. YOU ARE A HERO.
this got me over the hill I needed without digging up an Arduino Uno to deal with ISP