bdring/Grbl_Pen_Servo #2: Z0 - invalid (error: ID32) aftre G2

kirznervadim, 2019-11-25:
I use next code via LaserGRBL. And I got 'Z0 invalid gcode id: 32'
`Z0
G90
G21
G1 F1000
G1  X5.2951 Y5.4498
Z1 S90
G4 P1
G1 F1000.000000
G1  X5.2668 Y5.1497 F1000.000000
G2 X4.9869 Y5.0961 I-0.2159 J0.3698
G2 X4.5443 Y5.4938 I0.0809 J0.5352
G2 X4.5879 Y6.0662 I0.8499 J0.2232
G2 X5.3132 Y6.6525 I0.9226 J-0.3996
G2 X6.2199 Y6.5107 I0.2606 J-1.3027
G2 X6.9586 Y5.4619 I-0.7181 J-1.2905
G2 X6.7097 Y4.2168 I-1.7697 J-0.2935
G2 X5.3385 Y3.3223 I-1.6514 J1.0332
G2 X3.754 Y3.6816 I-0.3254 J2.2387
G2 X2.7023 Y5.3748 I1.3462 J2.0093
G2 X3.1733 Y7.2992 I2.7079 J0.357
G2 X5.1884 Y8.5089 I2.3658 J-1.6577
G2 X7.4528 Y7.9252 I0.3886 J-3.1769
G2 X8.8209 Y5.5884 I-1.9685 J-2.7214
G2 X8.1242 Y2.9839 I-3.6458 J-0.4202
G2 X5.4657 Y1.4569 I-3.0765 J2.2787
G2 X2.521 Y2.267 I-0.4519 J4.1146
G2 X0.8351 Y5.2472 I2.5884 J3.4312
G2 X1.7588 Y8.5321 I4.5832 J0.4837
G2 X5.0605 Y10.3771 I3.7857 J-2.8979
G2 X8.6858 Y9.3398 I0.5155 J-5.0517
G2 X10.69 Y5.7164 I-3.2072 J-4.1401
G2 X9.5388 Y1.7509 I-5.5202 J-0.5473
**Z0**
G4 P1
G1 F1000
G1 X0 Y0`

bdring, 2019-11-25:
**Z0** looks wrong

kirznervadim, 2019-11-25:
G-code was generated via Inkscape extension. 
in the settings of ext.: servoUP - Z1, servoDOWN - Z0

Why it's wrong?

bdring, 2019-11-25:
Sorry, impossible to edit markdown on my phone. It turned asterisks into bold. 

`**Z0**` is not valid gcode. Is that actually what was generated?

I formatted your message too.

kirznervadim, 2019-11-25:
I marked row which shows the Error: 32.
of course Z0 without any asterisks

bdring, 2019-11-25:
It is probably due to the machine being in G2 mode from the previous statement. It needs to be in G1 or G0 for a Z move like that. Place G0 in front

kirznervadim, 2019-11-26:
So, it's problem of Inkscape plugin which doesn't paste G0 after G2 , i'm right?

bdring, 2019-11-26:
Yes. 

G0 Z0

closed 2019-11-27 by kirznervadim