bdring/Grbl_Pen_Servo #4: Servos heating up? anybody?

eneene, 2021-01-08:
I noticed the servo is not disabled when the machine is stopped. It keeps making micro-adjustments. You can hear a faint grind when it is idle.
I can feel the servo getting warmer as well.
Could I use the spindle enable pin to control a transistor to cut vcc to the servo when the pin is low?
Or maybe there is a way to do that in the software?

eneene, 2021-01-08:
btw it absolutely works. But the spindle enable pin will go low before the pen go back to origin, so I used the stepper disable pin instead (digital pin 8).
In that case it is better to use a pnp bjt instead of an npn, because the pin will go high when the steppers are supposed to be disabled.

bdring, 2021-01-08:
You should make sure the servo has some free travel beyond the point you are stopping at. If it is hitting an endpoint, it can cause some overheating.

The servo should stop sending signals when the motors disable. I good quality servo should turn off with the signal. If you set $Stepper/IdleTime=nnn Where nnn is in milliseconds between 1 and 255. 255 is a special case and keeps the signal on.

You can also cancel the signal with $MD

https://github.com/bdring/Grbl_Esp32/wiki/Settings#motordisable-or-md

eneene, 2021-01-09:
Hey, thank you for the reply! I realized it was the rubber band i use to pull the pen down. It was making the servo fight to stay in place. The idle timer is a perfect solution, thank you very much.

closed 2021-01-09 by bdring