bdring/Grbl_Pen_Servo #7: are .STL still availeble?

hendrik607, 2022-04-03:
Sorry if i ask this in the wrong place. I would like to build this plotter, cannot find the STL files. Youtube text says "files for this plotter are now posted on my GitHub page (linked in the channel's "About" section) ". Maybe i search the wrong location. Thanks in advance!

gjs1947, 2022-09-12:
> Sorry if i ask this in the wrong place. I would like to build this plotter, cannot find the STL files. Youtube text says "files for this plotter are now posted on my GitHub page (linked in the channel's "About" section) ". Maybe i search the wrong location. Thanks in advance!

gjs1947, 2022-09-12:
????????????

gjs1947, 2022-09-12:
would like to build this plotter, cannot find the STL files. ????